stereolabs/zed-yolo #8: ZED>>ERROR:sl::Camera::Open has not been called,no Camera instance running

xrkwxy, 2019-03-24:
When I run the zed_cpp_sample,it has a error:ZED>>ERROR:sl::Camera::Open has not been called,no Camera instance running.I need help.Thank you!

LevineYang, 2019-07-18:
Same problem, did you figure it out?

cmrs2017, 2019-09-04:
ZED Explorer runs but the depth_sensing.py gives error
 ZED >> ERROR: sl::Camera::Open has not been called, no Camera instance running.

Also In Ros-wrapper, I  am able to subscribe rgb images but for disparity or depth topics, the zed node crashes.
Device: Tx2
sdk: 2.8.3
firmware:1523

VGrondin, 2019-10-18:
I had this error when launching ZEDfu in live mode :
ZED >> ERROR: sl::Camera::Open has not been called, no Camera instance running.
Solved it by using another USB 3.0 connector. The one in front wasn't working but when i plugged it at the back of the tower it worked !

cmrs2017, 2019-10-18:
Solved it. I think the issue was log file generation space. While running zed camera, it requires 3gb of free space on tx2. 
When the error came, tx2 was having around 1.2gb space only.

Amol-Kulkarni91, 2020-07-22:
I have enough space on tx2 and the usb 3.0 hub works. I'm still facing this issue though.

tlhcelik, 2020-08-05:
i receiving same error. 
I use 

- Xavier AGX 
- R32.3.1
- CUDA10.0
- 500 GB NVMe Disk
- OpenCV 4.1.1

ZED_Depth_Viewer is running but API is not working.

Code :
```
import pyzed.sl as sl

zed = sl.Camera()

init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.HD1080
init_params.camera_fps = 30

# Open the camera
err = zed.open(init_params)
if err != sl.ERROR_CODE.SUCCESS:
    exit(-1)
```

OUTPUT: 
**[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.**

saeedarabi92, 2020-08-05:
I also have this problem:
```
nvidia@nvidia-desktop:~$ roslaunch zed_display_rviz display_zed2.launch -v
... logging to /home/nvidia/.ros/log/11c1eeda-d74f-11ea-83c0-00044bc7359a/roslaunch-nvidia-desktop-15128.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

... loading XML file [/opt/ros/melodic/etc/ros/roscore.xml]
... executing command param [rosversion roslaunch]
Added parameter [/rosversion]
... executing command param [rosversion -d]
Added parameter [/rosdistro]
Added core node of type [rosout/rosout] in namespace [/]
... loading XML file [/home/nvidia/Desktop/Aria_navigation/src/zed-ros-examples/zed_display_rviz/launch/display_zed2.launch]
... loading XML file [/home/nvidia/Desktop/Aria_navigation/src/zed-ros-wrapper/zed_wrapper/launch/zed2.launch]
... loading XML file [/home/nvidia/Desktop/Aria_navigation/src/zed-ros-wrapper/zed_wrapper/launch/include/zed_camera.launch.xml]
... executing command param [/opt/ros/melodic/lib/xacro/xacro '/home/nvidia/Desktop/Aria_navigation/src/zed-ros-wrapper/zed_wrapper/urdf/zed_descr.urdf.xacro'                         camera_name:=zed2                         camera_model:=zed2                         base_frame:=base_link                         cam_pos_x:=0.0                         cam_pos_y:=0.0                         cam_pos_z:=0.0                         cam_roll:=0.0                         cam_pitch:=0.0                         cam_yaw:=0.0]
Added parameter [/zed2/zed2_description]
Added node of type [robot_state_publisher/robot_state_publisher] in namespace [/zed2/]
Added parameter [/zed2/zed_node/hue]
Added parameter [/zed2/zed_node/saturation]
Added parameter [/zed2/zed_node/auto_exposure_gain]
Added parameter [/zed2/zed_node/pos_tracking/path_max_count]
Added parameter [/zed2/zed_node/pos_tracking/publish_map_tf]
Added parameter [/zed2/zed_node/pos_tracking/floor_alignment]
Added parameter [/zed2/zed_node/pos_tracking/publish_tf]
Added parameter [/zed2/zed_node/pos_tracking/fixed_z_value]
Added parameter [/zed2/zed_node/pos_tracking/map_frame]
Added parameter [/zed2/zed_node/pos_tracking/path_pub_rate]
Added parameter [/zed2/zed_node/pos_tracking/odometry_frame]
Added parameter [/zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose]
Added parameter [/zed2/zed_node/pos_tracking/two_d_mode]
Added parameter [/zed2/zed_node/pos_tracking/area_memory]
Added parameter [/zed2/zed_node/pos_tracking/area_memory_db_path]
Added parameter [/zed2/zed_node/pos_tracking/initial_base_pose]
Added parameter [/zed2/zed_node/sharpness]
Added parameter [/zed2/zed_node/depth_texture_conf]
Added parameter [/zed2/zed_node/general/camera_flip]
Added parameter [/zed2/zed_node/general/verbose]
Added parameter [/zed2/zed_node/general/svo_compression]
Added parameter [/zed2/zed_node/general/zed_id]
Added parameter [/zed2/zed_node/general/camera_name]
Added parameter [/zed2/zed_node/general/gpu_id]
Added parameter [/zed2/zed_node/general/grab_frame_rate]
Added parameter [/zed2/zed_node/general/self_calib]
Added parameter [/zed2/zed_node/general/serial_number]
Added parameter [/zed2/zed_node/general/resolution]
Added parameter [/zed2/zed_node/general/base_frame]
Added parameter [/zed2/zed_node/video/img_downsample_factor]
Added parameter [/zed2/zed_node/video/extrinsic_in_camera_frame]
Added parameter [/zed2/zed_node/gain]
Added parameter [/zed2/zed_node/pub_frame_rate]
Added parameter [/zed2/zed_node/exposure]
Added parameter [/zed2/zed_node/point_cloud_freq]
Added parameter [/zed2/zed_node/brightness]
Added parameter [/zed2/zed_node/depth/depth_stabilization]
Added parameter [/zed2/zed_node/depth/depth_downsample_factor]
Added parameter [/zed2/zed_node/depth/openni_depth_mode]
Added parameter [/zed2/zed_node/depth/quality]
Added parameter [/zed2/zed_node/depth/sensing_mode]
Added parameter [/zed2/zed_node/whitebalance_temperature]
Added parameter [/zed2/zed_node/gamma]
Added parameter [/zed2/zed_node/auto_whitebalance]
Added parameter [/zed2/zed_node/mapping/fused_pointcloud_freq]
Added parameter [/zed2/zed_node/mapping/resolution]
Added parameter [/zed2/zed_node/mapping/max_mapping_range]
Added parameter [/zed2/zed_node/mapping/mapping_enabled]
Added parameter [/zed2/zed_node/depth_confidence]
Added parameter [/zed2/zed_node/contrast]
[... 164 lines cut ...]
[ INFO] [1596656876.785570763]:  * [DYN] sharpness		-> 4
[ INFO] [1596656876.786549676]:  * [DYN] gamma			-> 8
[ INFO] [1596656876.788344775]:  * [DYN] auto_exposure_gain	-> ENABLED
[ INFO] [1596656876.791723192]:  * [DYN] auto_whitebalance	-> ENABLED
[ INFO] [1596656876.807556198]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1596656876.807710187]:  *** Opening ZED 2...
[ INFO] [1596656876.867172741]: rviz version 1.13.7
[ INFO] [1596656876.867326122]: compiled against Qt version 5.9.5
[ INFO] [1596656876.867376747]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1596656876.892437965]: Forcing OpenGl version 0.
[ INFO] [1596656877.980069006]: Stereo is NOT SUPPORTED
[ INFO] [1596656877.980379000]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1596656898.581507658]: ZED connection -> LOW USB BANDWIDTH
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1596656915.555467661]: ZED connection -> LOW USB BANDWIDTH
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1596656936.675089604]: ZED connection -> LOW USB BANDWIDTH
[... cut ...] era::Open has not been called, no Camera instance running.
[ INFO] [1596656962.111657962]: ZED connection -> LOW USB BANDWIDTH
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1596656981.833006756]: ZED connection -> LOW USB BANDWIDTH
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1596657004.466018711]: ZED connection -> LOW USB BANDWIDTH
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
```

closed 2022-04-26 by github-actions[bot]