stereolabs/zed-yolo #10: darknet.sln fails to build

Docker300, 2019-05-14:
I am following these instructions (https://github.com/stereolabs/zed-yolo/tree/master/libdarknet)   to compile darknet for Zed Yolo on Windows? It is said that 

> If you have MSVS 2015, CUDA 9.1, cuDNN 7.0 and OpenCV 3.x (with paths: C:\opencv_3.0\opencv\build\include & C:\opencv_3.0\opencv\build\x64\vc14\lib), then start MSVS, open build\darknet\darknet.sln, set x64 and Release https://hsto.org/webt/uh/fk/-e/uhfk-eb0q-hwd9hsxhrikbokd6u.jpeg and do the: Build -> Build darknet. 

When I open "zed-yolo-master\libdarknet\build\darknet\darknet.sln" in Visual Studio Community 2015 it fails to load darknet.sln

[image]

So I tried to use Visual studio 2017 and 2019. Although, It loads darknet in VS 2017 and 2019. I included the opencv path as descried but when I build it  (Build -> Build Solution),  It fails to build as shown here in following figure.

[image]

Can any one guide me that what is the reason of this problem?

AlexeyAB, 2019-05-14:
@Docker300 You can try to build this repository https://github.com/AlexeyAB/darknet by using Cmake, there will be created `uselib.exe` that uses ZED-camera:
`uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`

Docker300, 2019-05-19:
I am unable to find uselib.exe file in darknet folder when I compiled this repository according to the instructions given in How to compile on Windows (using vcpkg)

> If you have already installed Visual Studio 2015/2017/2019, CUDA > 10.0, cuDNN > 7.0, OpenCV > 2.4, then compile Darknet by using C:\Program Files\CMake\bin\cmake-gui.exe as on this IMAGE: Configure -> Optional platform for generator (Set: x64) -> Finish -> Generate -> Open Project -> x64 & Release -> Build -> Build solution

I have Visual Studio 2017, CUDA 10.0, cuDNN  7.5, OpenCV 3.4.0, CMAKE 3.13.4

When I use uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera
it gives following error 
> 'uselib.exe' is not recognized as an internal or external command,
operable program or batch file.
Where is this file located?

AlexeyAB, 2019-05-19:
If you compiled with **x64 & Release** https://hsto.org/webt/uh/fk/-e/uhfk-eb0q-hwd9hsxhrikbokd6u.jpeg
then `uselib.exe` will be created in the `Release` folder.

Docker300, 2019-05-19:
Thank You for the help
I am newbie.  I compiled Darknet.sln with x64 & Release. and when I use  

> uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera

 in command prompt it give error

> uselib.exe System error 

> pthreadVC2.dll does not exist so the code execution can not proceed. So reinstall the program can resolve this issue.

[image]

I don't know why is it giving this error because project is successfully build
 
[image]

But pthread.h is also not found in CMAKE. 
[image]

Would you please guide me ?
Am I getting uselib.exe-System error because pthread.h is not found in CMAKE? 
If it is the reason then how can I solve it?
Do I need to re-install the CMAKE? or Do I need to make any change in any file to solve this issue?

>

AlexeyAB, 2019-05-19:
1. Copy `darklib.dll` and `uselib.exe` from `darknet/Release` folder to the `darknet/` root folder.

2. Also copy `pthreadGC2.dll` and `pthreadVC2.dll` from `darknet\3rdparty\pthreads\bin\` folder to the `darknet/` root folder.

3. Download https://pjreddie.com/media/files/yolov3.weights to the `darknet/` root folder.

4. Then run `uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`
    **in the `darknet/` root folder.**

Docker300, 2019-05-21:
Thank You for the detailed reply :) 

When I run uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera
I get this error

> input image or video filename: **ZED 3D Camera NOT ENOUGH GPU MEMORY**
>  Error: ZED Camera should be connected to USB 3.0. And ZED_SDK should be installed.

But ZED camera is connected and ZED SDK is also installed in  C:\Program Files (x86)\ZED SDK 
(CMAKE has also detected ZED SDK at the same location C:\Program Files (x86)\ZED SDK)

I do not get this error when I run ZED SDK Example projects.
But I get this error when I run ZED-YOLO.

Would you please tell me that why am I getting  
> ZED 3D Camera NOT ENOUGH GPU MEMORY

error and how should I solve it?

I am using Nvidia GeForce 840M and latest version of ZED SDK 2.8

[image]
.
.
.
[image]

AlexeyAB, 2019-05-21:
It seems that you don't have enough GPU-RAM to run full yolov3-model.

Try to run tiny model:

* Download: https://pjreddie.com/media/files/yolov3-tiny.weights

* run: 

`uselib.exe data/coco.names cfg/yolov3-tiny.cfg yolov3-tiny.weights zed_camera`

Docker300, 2019-05-23:
Thank You.  Although, It is not showing "Not enough GPU" error now.
But it gives another error "ZED 3D Camera Failed to setup" and stops when I run either Yolo3 or Yolo-Tiny

I don't know that why it is giving this error .
For testing the camera, I opened ZED explorer to check if camera is setup or not. In ZED explorer it opens the camera, it means there is no issue related to camera setup. Then why it is giving this error here when I run it for YOLO 

Please guide me
Do I need to make some changes in any file?

[image]

AlexeyAB, 2019-05-23:
Try to un-comment this line and recompile: https://github.com/AlexeyAB/darknet/blob/9c9344a1ff4541499c8f69ea997be5145b4e1de3/src/yolo_console_dll.cpp#L340

What GPU do you use?

Docker300, 2019-05-24:
Line# 340 in yolo_consol_dll.cpp is already commented.
I am using Nvidia GeForce 840M (GPU).

AlexeyAB, 2019-05-24:
Un-comment, i.e. change this line:
`//init_params.sdk_cuda_ctx = (CUcontext)detector.get_cuda_context();`
to this
`init_params.sdk_cuda_ctx = (CUcontext)detector.get_cuda_context();`

Docker300, 2019-05-24:
I did it but facing the same issue :(

Docker300, 2019-06-03:
I am getting this error and tried to fix it many times.

>  ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1
> CAMERA FAILED TO SETUP

Although, It build the project successfully but does not show any output due to this error.
I am facing this problem not only in YOLO but also in all other ZED examples now.  Problem is still there after reinstalling the SKD
Would you please guide me?

AlexeyAB, 2019-06-03:
@Docker300 Try to recompile Darknet Yolo

Docker300, 2019-06-03:
I tried every possible solution except reinstalling windows... 
After re-installing Zed-SDK I also re-installed cuda 10.0 ( after deleting its old installation)
 I recomiled Darknet YOLO but getting same message 
[image]

When I run C:\Program Files (x86)\ZED SDK\tools\ZED Depth Viewer.exe
It shows the camera.
But
I don't know that why it fails to start camera when I run the code either darknet-YOLO or Zed master examples.             
I have been trying to solve this issue since last two weeks.
Please suggest me some solution.

AlexeyAB, 2019-06-03:
> ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1
> CAMERA FAILED TO SETUP
> 

What is the current SDK version do you use?

It seems that you try to run old SDK (2.8.0) exe-files

So it seems that somewhere there are still old and new versions of ZED SDK 

1. Try to delete all files of old SDK

2. Also try to delete your Darknet, download it again and compile again.

Docker300, 2019-06-04:
I did all these steps and I'm using SDK 2.8.1 now.
After deleting SDK 2.8.0 it does NOT give message  

> ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1

But 

> CAMERA FAILED TO SETUP

problem is still there when I run ZED Examples.

Also, gives the same message message with I run YOLO after downloading and compiling it.
Although, it does not give any error in build
But I don't know that whey it fails to run camera in code or SDK
while camera is successfully detected by zed explorer.exe

Docker300, 2019-06-07:
Thank  You, It is working now.
Would you please tell me that 
To train ZED-YOLO 3D ( with depth estimation for 3D object detection)

- Do I have to use labeled 3D image dataset or labeled 2D image dataset?
 Or
- Labeled dataset of both 2D images and  labeled dataset of its Depth images?

AlexeyAB, 2019-06-07:
You should use 2D labels as usual. ZED will estimate 3D coords automatically.

Docker300, 2019-06-07:
Okay.
Thank You :)

Docker300, 2019-06-10:
Same command is used to train the yolo-tiny with depth estimation for ZED?
> darknet.exe detector train data/obj.data yolov3-tiny.cfg darknet53.conv.74
 Or
we have to use different command?

AlexeyAB, 2019-06-10:
Read: https://github.com/AlexeyAB/darknet#how-to-train-tiny-yolo-to-detect-your-custom-objects

Docker300, 2019-06-12:
Please guide me
In YOLO-ZED
What is **_x_** and **_y_**?
Is it height and width of bounding box?
and How **z** is calculated?
Does ZED_YOLO extracts the distance or **_z_** (depth) value by taking the center of bounding box(width/2, height/2) of detected image?

adujardin, 2019-06-12:
The current ZED Yolo simply takes the 2D detector output and compute the 3D world coordinates from the depth map computing at runtime during the inference. 

The training phase doesn't take into account the 3D at all. It's just a regular Yolo 2D detector

Docker300, 2019-06-12:
Okay. Thank You
Would you please explain about x, y and z value here?
 _**x**_, _**y**_ values indicate the width and height of bound box  while **_z_** shows the distance from camera? 

[image]

AlexeyAB, 2019-06-12:
@Docker300 `x, y, z` - are 3D coordinates of center of object (measured in meters relative to left camera)

Docker300, 2019-06-13:
Okay. Thank You

Docker300, 2019-06-18:
For training
1. Is it okay If image size is 450 X 600 (instead of 450 x 450 or 600 x 600) ?
2. In this case, should we also change  the width and height in .cfg file according to our image size?
3. if Yes then should we change other parameters according to 450 x 600 image size?

Docker300, 2019-06-19:
I have trained YOLO on my own dataset.  I have to use zed camera. For this, when I use uselib.exe.
On test image and video files, **it does not show labels and depth info when I run uselibe.exe** it only draws bounding boxes
[image]

But
When I run  darknet.exe it shows the bounding boxes with labels like this
[image]

What would be the issue?

AlexeyAB, 2019-06-19:
Because you use wrong names-file in your command:
`uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`

It can show depth info only on **Video**-stream from ZED-camera.

Docker300, 2019-06-20:
You are right. Thank You  :)

umair-nazeer, 2019-07-27:
I  can't able to use wingpudll compiled by visual studio the file do exists at path and os.exists also returns true but CDLL is not able to open the dll picture is attached please check
[image]

Whisper94, 2020-04-30:
@AlexeyAB
Hello. If I try to use this command `uselib.exe data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera` then I'll get following error 
[image]

Bevor I've copied opencv_world3410.dll from opencv-folder because it couldn't be found from darknet.

Thank you!

ibrahimaa9, 2020-05-16:
why it doesn't work 
[image]

leemaru, 2020-07-16:
> Thank You, It is working now.
> Would you please tell me that
> To train ZED-YOLO 3D ( with depth estimation for 3D object detection)
> 
> * Do I have to use labeled 3D image dataset or labeled 2D image dataset?
>   Or
> * Labeled dataset of both 2D images and  labeled dataset of its Depth images?

How did you get this to work? I am having the same error. But I am on a brand new PC with RTX2070, so I know I have enough RAM and the proper GPU.

FatemehEsfahani, 2021-01-11:
I get pthreadvc2.dll not found. How I can fix it?

closed 2022-04-26 by github-actions[bot]