stereolabs/zed-yolo #11: Video-stream stopped! error

SS47816, 2019-05-17:
when running `./darknet detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights zed_camera`,

encountered error: `Video-steam stopped!`

AlexeyAB, 2019-05-17:
compile this repository https://github.com/AlexeyAB/darknet by using 
```
cmake .
make
```

and run 
`./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`

or
`LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`

SS47816, 2019-05-21:
I re-downloaded the darknet repo, set `GPU=1`, `cuDNN=1`, `opencv=1`, `ZED camera=1`, and ran: `cmake .` and `make`

now a new issue occurred: 

```bash
[ 41%] Building CUDA object CMakeFiles/darklib.dir/src/activation_kernels.cu.o
nvcc fatal   : Unknown option 'fopenmp'
CMakeFiles/darklib.dir/build.make:1646: recipe for target 'CMakeFiles/darklib.dir/src/activation_kernels.cu.o' failed
make[2]: *** [CMakeFiles/darklib.dir/src/activation_kernels.cu.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/darklib.dir/all' failed
make[1]: *** [CMakeFiles/darklib.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```
Do you know how I can solve this?

AlexeyAB, 2019-05-21:
What CUDA version do you use?

Can you show output of command?
```
nvcc --version
gcc --version
```

SS47816, 2019-05-23:
Here are my outputs:

```bash
$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sun_Sep_30_21:09:22_CDT_2018
Cuda compilation tools, release 10.0, V10.0.166

$ gcc --version
gcc (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

BTW, is it possibly because that I am using a Jetson TX2?

Thanks a lot for the help :smile:

adujardin, 2019-06-14:
The command in the README was incorrect. It's indeed `uselib` and not the `darknet` command that should be used with the ZED, as described in Alexey's comment.

It has been fixed (https://github.com/stereolabs/zed-yolo/commit/17dbeec45c9827b524e162468db3d8f26553af06)

joseware, 2019-11-25:
Hi, I have the same error running on EC2 aws,
Video-steam stopped!

Training works ok but detection no

Help

Lwishh, 2020-01-13:
If you are using Windows and getting error after processing "Video-steam stopped!" you should check what you wrote in Command Promp.

Notation should be like:  
darknet.exe detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights data/<videoname>.mp4

For example:
darknet.exe detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights data/cars.mp4

prateekgulati, 2020-02-17:
Your path of video is incorrect. Either place the video in darknet folder or provide the correct path for the video on which you want the detection to work.

monowang, 2020-07-13:
> Hi, I have the same error running on EC2 aws,
> Video-steam stopped!
> 
> Training works ok but detection no
> 
> Help

Confirmed. Same on aws Sagemaker. Works for images, but failed with video.

Everything works in colab as well as local machines. 

PS: double checked the path.

AruparnaMaity, 2020-11-01:
> > I re-downloaded the darknet repo, set GPU=1, cuDNN=1, opencv=1, ZED camera=1, and ran:
> > cmake .
> > make
> 
> Can you try to re-downloaded the darknet repo again, set GPU=1, cuDNN=1, opencv=1, ZED camera=1 in the Makefile
> and do
> `make`
> instead of
> 
> ```
> cmake .
> make
> ```
> 
> then run
> `./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`
> 
> or
> `LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera`

This worked for me. Thanks.

MakeWater, 2021-04-21:
I counter the problem when I run "./darknet detector demo data/jueyuanzi.data yolov3.cfg jueyuanzi_backup/yolov3_best.weights --ext_output data/testvideo/jyz_hxq_004.mp4", I followed AlexyAB advise to run "cmake ." and "sudo make" and run the command still has the same problem. How it''s turn out that there only one "-" in frount of "ext_output" option! 
Yeah,  "--ext_output" is wrong while "-ext_output" work fine.

MakeWater, 2021-04-21:
I encounter the problem when I run "./darknet detector demo data/jueyuanzi.data yolov3.cfg jueyuanzi_backup/yolov3_best.weights --ext_output data/testvideo/jyz_hxq_004.mp4", I followed AlexyAB advise to run "cmake ." and "sudo make" and run the command still has the same problem. How it''s turn out that there only one "-" in frount of "ext_output" option! 
Yeah,  "--ext_output" is wrong while "-ext_output" work fine.

closed 2022-04-26 by github-actions[bot]

devilshark, 2022-12-23:
Hello.

I have the samed issue. I have used this code serveral months ago and it'was working
`darknet.exe detector demo data/creo.data yolo-creo.cfg yolo-creo_4000.weights -ext_output video/poissons.mp4`

But now I've a "Video-stream stopped".

I try to change the name of my video and I've the same error... May be my video is not in the good repository but how to know ? I try to download the new OpenCv released but no change.

Thanks for your help