stereolabs/zed-yolo #17: Video-stream stopped! error on running videos

yudhiApp, 2019-08-01:
Hello,
I am using a remote server with gpu. The AlexeyAB repository works for me on images only:

./darknet detector test ./cfg/coco.data ./cfg/yolov3.cfg ./yolov3.weights data/dog.jpg

When I tried to run it on a video with **opencv/3.4.3-video** I get video-stream stopped error which never stops and when I use **opencv/3.4.1-cuda-9.1**, it runs but does not save the video as res.avi (I do not see cvWriteFrame)

./darknet detector demo ./cfg/coco.data ./cfg/yolov3.cfg yolov3.weights ./data/street.mp4 -dont_show -out_filename res.avi

I have these : cuda/9.1-cudnn7.0.5 ( I tried cuda 10.0, same problem), gcc/6.5.0

what should I do? Thanks

Gazi13, 2020-02-13:
any solution ?

prateekgulati, 2020-02-17:
Your path of video is incorrect. Either place the video in darknet folder or provide the correct path for the video on which you want the detection to work.

AdityaLoth, 2020-05-17:
facing the same error and no solution works for me.

itsjustmithun, 2020-05-26:
Hey, facing the same issue.
Works just fine with images, but throws up video-stream stopped error for videos. Did anyone got it working?

Gazi13, 2020-05-26:
> Your path of video is incorrect. Either place the video in darknet folder or provide the correct path for the video on which you want the detection to work.

This worked for me and i am using this for videos. Just change the path. 
Also this command for google colab,

!./darknet detector demo data/obj.data cfg/yolo-car_train.cfg backup/yolo-car_train_final.weights outputVideo-23.avi -out_filename outputVideo/carVideoResult-23.mp4 -dont_show -map

itsjustmithun, 2020-05-27:
> This worked for me and i am using this for videos. Just change the path.
> Also this command for google colab,
>
> !./darknet detector demo data/obj.data cfg/yolo-car_train.cfg [...]

Hey, I have my current video file inside darknet folder

`C:\Packages\Yolo `V4\darknet\build\darknet\x64`

and I am using this command to test the video file

`darknet.exe detector demo cfg/coco.data cfg/yolov4.cfg yolov4.weights test.mp4`

where am I going wrong, can you please direct me. Thank you

monowang, 2020-07-13:
> Your path of video is incorrect. Either place the video in darknet folder or provide the correct path for the video on which you want the detection to work.

I am aware of what you suggest, that was also my first guess. But after carefully checking the path, I guess there is something difference in aws sagemaker data storage, since it works in colab.

Anay2805, 2020-08-28:
name as test.mp4
and dont give space in video name

AmnaFish, 2020-09-15:
Something that worked for me was that i used .avi format of video, instead of .mp4 because it was not working. It could be because im using CPU and not GPU or CUDA.
So my command was,
./darknet detector demo data/coco.data cfg/yolov4-custom.cfg yolov4-custom.weights -dont_show data/video.avi -out_filename result.avi

JasonHarambe, 2020-10-17:
so i tried to put it in /content/drive/My Drive/test.mp4 - it shows Video-Stream stopped. However, I copied the file into /content/test.mp4 - it worked.

Roopana, 2020-10-26:
I am running YOLO4 on colab and faced teh same issue. Looks like the default directory is darknet. I copied video to data folder and ran using below command and it worked fine without the error:

!./darknet detector demo cfg/coco.data cfg/yolov4.cfg yolov4.weights data/test.mp4 -i 0 -out_filename results.avi

ritz-lab, 2021-01-06:
> When I tried to run it on a video with **opencv/3.4.3-video** 

How are you doing this? I'm trying to make the algorithm run on the IP of a video stream, by RTMP protocol, and I'm using the command:

`./darknet detector test ./cfg/coco.data ./cfg/yolov3.cfg ./yolov3.weights rtmp://IP_ADDRESS`

but I think the link is not being reached :( pls help

Khazina-python, 2021-03-06:
I was facing the same issue but now I fixed it. 
Here's what I did. 
For test video, you can upload it from your pc.Here I uploaded a test video named test.mp4.  So for that you use the command:

**%cd ..
upload()
%cd darknet**

Then in next step, use this command, making sure that the name of the video (mp3 or avi format) matches with your uploaded video name.

**!./darknet detector demo data/obj.data cfg/yolov4-obj.cfg /mydrive/yolov4/backup/yolov4-obj_best.weights -dont_show test.mp4 -i 0 -out_filename results.avi**

Then it will start to process it. After the processing is done, you can download your detected video file in avi format or mp4

**download('results.avi')**

Risingabhi, 2021-04-07:
There is a chance , this solution will work,
%cd /content/darknet
!sed -i 's/OPENCV=0/OPENCV=1/' Makefile
!sed -i 's/GPU=0/GPU=1/' Makefile
!sed -i 's/CUDNN=0/CUDNN=1/' Makefile
!sed -i 's/CUDNN_HALF=0/CUDNN_HALF=1/' Makefile
!apt update
!apt-get install libopencv-dev

yareenm, 2021-12-04:
Hey,
I'm facing the same issue. I have my current video file inside darknet folder

C:\Packages\Yolo V4\darknet\build\darknet\x64

and I am using this command to test the video file

darknet.exe detector demo cfg/coco.data cfg/yolov4.cfg yolov4.weights walk.mp4

The layers are loading successfully but after I'm getting error video-stream stopped. Can someone help me? Thanks.

Pepeter123, 2021-12-29:
Hello,

I found out sollution on this problem by adding flag `-ext_output`
In the example below i used my own mp4 file I've named test
`./darknet detector demo cfg/coco.data cfg/yolov4.cfg yolov4.weights -ext_output test.mp4`

More option you can find on
https://github.com/AlexeyAB/darknet/#how-to-use-on-the-command-line

Best regards

devofficer, 2022-03-22:
Hello,
Your problem is neither video path nor incorrect command.
I had that issue and worked out for several days, ending with finding out right solution to it.

So here comes my solution:
- Please install latest opencv. (https://sourceforge.net/projects/opencvlibrary/files/4.5.5/opencv-4.5.5-vc14_vc15.exe/download)
- Add your "bin" folder path of opencv to the environment variable path. (You may move extracted opencv folder into where you prefer)

And try running test on a video file!
Hope it works.

closed 2022-04-27 by github-actions[bot]