stereolabs/zed-yolo #24: What is the principle of this project?

willzoe, 2020-01-16:
I want to know how the distance information is determined in the detection results? Is there a block diagram or an article describing the relevant algorithm?

Tetsujinfr, 2020-08-29:
I am no representative of Stereolabs, just a user. Given the short description in the readme 

> "The left image will be used to display the detected objects alongside the distance of each, using the ZED Depth."

 I assume, for instance, they can average the per pixel depth estimation of the ZED camera of the area of the rectangle class for a given object (maybe excluding overlapping area from other objects) from Yolo. The ZED SDK is no opensource software so again, only assumptions here.

Tetsujinfr, 2020-08-29:
... some details I found on how the depth estimation is done along the yolo model [here](https://github.com/stereolabs/zed-yolo/issues/21)
pls consider closing the issue if that looks good enough to you.

closed 2022-04-26 by github-actions[bot]