stereolabs/zed-yolo #26: Error in sl/Camera.hpp

JohnieBraaf, 2020-02-20:
I'm unable to compile darknet with ZED SDK 3.0.

It seems that GPU is already defined somewhere.

/usr/local/zed/include/sl/Camera.hpp: At global scope:
<command-line>:0:5: error: expected identifier before numeric constant
/usr/local/zed/include/sl/Camera.hpp:1952:9: note: in expansion of macro ?GPU?
         GPU  /**< GPU Memory (Graphic card side).*/

adujardin, 2020-02-21:
Darknet defines a GPU macro globally. By default since this is a macro, it replaces this enum value also. An undef was added [here](https://github.com/AlexeyAB/darknet/blob/1de03f948cf860e18f6ad67bbfa68147297f0754/src/yolo_console_dll.cpp#L32) to try to limit this issue.

It seems it may not be enough sometimes, some users have reported this. As a workaround, you could add this `#undef GPU` directly into Camera.hpp for now (before the `MEM` enum).

We'll probably add this to the SDK if there are no side effects.

JohnieBraaf, 2020-02-21:
Thnx for pointing me in the right direction.
There were a couple of other small bits that needed adjustment. See below.

darknet/src/yolo_console_dll.cpp:99

cv::Mat slMat2cvMat(sl::Mat &input) {
    // Mapping between MAT_TYPE and CV_TYPE
    int cv_type = -1;
    switch (input.getDataType()) {
    case sl::MAT_TYPE**::F32**_C1:
        cv_type = CV_32FC1;
        break;
    case sl::MAT_TYPE**::F32**_C2:
        cv_type = CV_32FC2;
        break;
    case sl::MAT_TYPE**::F32**_C3:
        cv_type = CV_32FC3;
        break;
    case sl::MAT_TYPE**::F32**_C4:
        cv_type = CV_32FC4;
        break;
    case sl::MAT_TYPE**::U8**_C1:
        cv_type = CV_8UC1;
        break;
    case sl::MAT_TYPE**::U8**_C2:
        cv_type = CV_8UC2;
        break;
    case sl::MAT_TYPE**::U8**_C3:
        cv_type = CV_8UC3;
        break;
    case sl::MAT_TYPE**::U8**_C4:
        cv_type = CV_8UC4;
        break;
    default:
        break;
    }
    return cv::Mat(input.getHeight(), input.getWidth(), cv_type, input.getPtr<sl::uchar1>(sl::MEM**::**CPU));
}

cv::Mat zed_capture_3d(sl::Camera &zed) {
    sl::Mat cur_cloud;
    zed.retrieveMeasure(cur_cloud, sl::MEASURE**::XYZ**);
    return slMat2cvMat(cur_cloud).clone();
}

Line 335

#ifdef ZED_STEREO
                sl::InitParameters init_params;
                init_params.depth_minimum_distance = 0.5;
                init_params.depth_mode = sl::DEPTH_MODE**::ULTRA**;
                init_params.camera_resolution = sl::RESOLUTION**::HD720**;// sl::RESOLUTION_HD1080, sl::RESOLUTION_HD720
                init_params.coordinate_units = sl::UNIT::METER;
                //init_params.sdk_cuda_ctx = (CUcontext)detector.get_cuda_context();
                init_params.sdk_gpu_id = detector.cur_gpu_id;
                **//init_params.camera_buffer_count_linux = 2;
                //if (file_ext == "svo") init_params.svo_input_filename.set(filename.c_str());**
                if (filename == "zed_camera" || file_ext == "svo") {
                    std::cout << "ZED 3D Camera " << zed.open(init_params) << std::endl;
                    if (!zed.isOpened()) {
                        std::cout << " Error: ZED Camera should be connected to USB 3.0. And ZED_SDK should be installed. \n";
                        getchar();
                        return 0;
                    }
                    cur_frame = zed_capture_rgb(zed);
                    use_zed_camera = true;
                }
#endif  // ZED_STEREO

Line 410:

 if (use_zed_camera) {
                            while (zed.grab() != **sl::ERROR_CODE::SUCCESS**) std::this_thread::sleep_for(std::chrono::milliseconds(2));
                            detection_data.cap_frame = zed_capture_rgb(zed);
                            detection_data.zed_cloud = zed_capture_3d(zed);
                        }

I'm now able to run, but still not able to run with command (might be an issue with my OpenCV though:
LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera

Output:
input image or video filename: [ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (711) open OpenCV | GStreamer warning: Error opening bin: **no element "zed_camera"**

Does this issue need to be opened in the darknet repositiory?

adujardin, 2020-02-24:
You should pull the latest master, [this PR](https://github.com/AlexeyAB/darknet/pull/4762) was merged recently and includes the SDK 3.0 compatibility fixes.

JohnieBraaf, 2020-03-03:
Thnx, that fixed the build issues.

It is still needed to apply #undef GPU though.

closed 2020-03-03 by JohnieBraaf