stereolabs/zed-yolo #29: CUDA status Error: file: ./src/convolutional_kernels.cu : ()

ArghyaChatterjee, 2020-04-16:
I have Nvidia GPU-1070M, Graphics Driver-440.83, Cuda-10.2, Cudnn-7.6.5 & OpenCv-3.4.8. I get the following error during running:
```
arghya@arghya-Erazer-X7849-MD60379:~/darknet$ python3 darknet_zed.py
INFO:__main__:Opening ZED Camera...
 Try to load cfg: ../darknet/cfg/yolov3-tiny.cfg, weights: yolov3-tiny.weights, clear = 0 
 compute_capability = 610, cudnn_half = 0 
net.optimized_memory = 0 
mini_batch = 1, batch = 1, time_steps = 1, train = 0 
   layer   filters  size/strd(dil)      input                output
   0 conv     16       3 x 3/ 1    416 x 416 x   3 ->  416 x 416 x  16 0.150 BF
   1 max                2x 2/ 2    416 x 416 x  16 ->  208 x 208 x  16 0.003 BF
   2 conv     32       3 x 3/ 1    208 x 208 x  16 ->  208 x 208 x  32 0.399 BF
   3 max                2x 2/ 2    208 x 208 x  32 ->  104 x 104 x  32 0.001 BF
   4 conv     64       3 x 3/ 1    104 x 104 x  32 ->  104 x 104 x  64 0.399 BF
   5 max                2x 2/ 2    104 x 104 x  64 ->   52 x  52 x  64 0.001 BF
   6 conv    128       3 x 3/ 1     52 x  52 x  64 ->   52 x  52 x 128 0.399 BF
   7 max                2x 2/ 2     52 x  52 x 128 ->   26 x  26 x 128 0.000 BF
   8 conv    256       3 x 3/ 1     26 x  26 x 128 ->   26 x  26 x 256 0.399 BF
   9 max                2x 2/ 2     26 x  26 x 256 ->   13 x  13 x 256 0.000 BF
  10 conv    512       3 x 3/ 1     13 x  13 x 256 ->   13 x  13 x 512 0.399 BF
  11 max                2x 2/ 1     13 x  13 x 512 ->   13 x  13 x 512 0.000 BF
  12 conv   1024       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x1024 1.595 BF
  13 conv    256       1 x 1/ 1     13 x  13 x1024 ->   13 x  13 x 256 0.089 BF
  14 conv    512       3 x 3/ 1     13 x  13 x 256 ->   13 x  13 x 512 0.399 BF
  15 conv    255       1 x 1/ 1     13 x  13 x 512 ->   13 x  13 x 255 0.044 BF
  16 yolo
[yolo] params: iou loss: mse (2), iou_norm: 0.75, cls_norm: 1.00, scale_x_y: 1.00
  17 route  13 		                           ->   13 x  13 x 256 
  18 conv    128       1 x 1/ 1     13 x  13 x 256 ->   13 x  13 x 128 0.011 BF
  19 upsample                 2x    13 x  13 x 128 ->   26 x  26 x 128
  20 route  19 8 	                           ->   26 x  26 x 384 
  21 conv    256       3 x 3/ 1     26 x  26 x 384 ->   26 x  26 x 256 1.196 BF
  22 conv    255       1 x 1/ 1     26 x  26 x 256 ->   26 x  26 x 255 0.088 BF
  23 yolo
[yolo] params: iou loss: mse (2), iou_norm: 0.75, cls_norm: 1.00, scale_x_y: 1.00
Total BFLOPS 5.571 
avg_outputs = 341534 
 Allocate additional workspace_size = 52.43 MB 
 Try to load weights: yolov3-tiny.weights 
Loading weights from yolov3-tiny.weights...
 seen 64, trained: 32013 K-images (500 Kilo-batches_64) 
CUDA status Error: file: ./src/convolutional_kernels.cu : () : line: 138 : build time: Apr 16 2020 - 20:44:28 

 CUDA Error: invalid resource handle
python3: : Unknown error 864085283
```
Can anybody help regarding the matter?

patthongtra, 2020-09-08:
I got the same problem. Anyone have solutions?

zhlhenry113, 2020-09-16:
Did you fix that? I also have the same problem

xiayanqian, 2020-10-26:
what I do:
1. cd in the folder: ~/zed-yolo/zed_python_sample
2. Use the full command: python3 darknet_zed.py -c cfg/yolov3-tiny.cfg -w yolov3-tiny.weights -m cfg/coco.data -t 0.5
3. According to the errors, create cfg/ folder, data/ folder and copy the yolov3-tiny.cfg, yolov3-tiny.weights, coco.data, coco.name files to their position
4. Use the command, and no CUDA Error: invalid resource handle occurs.

closed 2022-04-26 by github-actions[bot]