stereolabs/zed-yolo #33: ValueError: NULL pointer access

husseinpasha, 2020-08-14:
Hi everyone,
I am trying to run this program on Jetson TX2. The Jetpack version is 4.4 and I haven't installed OpenCV on the Jetson from source. I have no problem detecting the images. Also _darkent_video.py_ works. But whenever I try to run  _darknet_zed.py_ I get the below error. I use yolov3. While when I use ylov3-tiny weights the programs works for a few seconds and then gives the same error again.

Traceback (most recent call last):
  File "darknet_zed.py", line 482, in <module>
    main(sys.argv[1:])
  File "darknet_zed.py", line 441, in main
    detections = detect(netMain, metaMain, image, thresh)
  File "darknet_zed.py", line 247, in detect
    if dets[j].prob[i] > 0:
ValueError: NULL pointer access

Finally when I run _./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera_ I get below error:

 CUDA Error: invalid argument
CUDA Error: invalid argument: Success
uselib: ./src/utils.c:326: error: Assertion `0' failed.
Aborted (core dumped)
jetson@jetson-desktop

Shame-fight, 2020-09-15:
For the first point, I have the same error in ubuntu16.04, CUDA10, zed_sdk3.2.2. when I use ylov3-tiny weights the programs works for a few seconds and gives the same error:
`INFO:__main__:person: 33.0%
INFO:__main__:FPS: 38.78371830673349
Traceback (most recent call last):
  File "darknet_zed.py", line 479, in <module>
    main(sys.argv[1:])
  File "darknet_zed.py", line 438, in main
    detections = detect(netMain, metaMain, image, thresh)
  File "darknet_zed.py", line 246, in detect
    if dets[j].prob[i] > 0:
ValueError: NULL pointer access
`
I use the ./uselib... command without any problem, and yolov3-tiny can run normally.
By the way: Can I use zed on usb2.0, is there any way to do it?
@husseinpasha @Poutsh @nesnes @adujardin @obraun-sl

Shame-fight, 2020-09-15:
This problem occurs when a person is detected

husseinpasha, 2020-09-16:
@Shame-fight , Same for me! Actually ylov3-tiny just detects the person and as soon as it does that the program shuts down!

closed 2020-09-16 by husseinpasha

rykymai, 2020-10-23:
Did you solve it?

caffreyu, 2020-12-09:
I got the same NULL pointer access error and the same error message. Does anyone have some clues about it?

niemiaszek, 2020-12-21:
@caffreyu It is solved by pull #37