stereolabs/zed-yolo #36: ValueError: NULL pointer access when object is detected

rykymai, 2020-10-28:
[image]
I have this issuse every time it detects a person, i'm testing it with yolov3-tiny, if i run instead ./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera it works

rykymai, 2020-11-18:
solved

closed 2020-11-18 by rykymai

niemiaszek, 2020-12-21:
@rykymai how did you solve this?

niemiaszek, 2020-12-21:
Okay, I solved it myself. Next time, please, explain the solution, to help people facing the same problem. Solution is provided by AlexeyAB [there](https://github.com/AlexeyAB/darknet/issues/6322#issuecomment-663215126). You have to change class DETECTION to new format. One used inn darknet_zed.py is outdated. I will prepere pull request, hopefully someone will notice it.

yuxuan-wang-tj, 2021-11-24:
@niemiaszek I can't solve this problem, could you give me some details ?

niemiaszek, 2021-11-24:
Do you have new version of this repository with my Pull Request? It solved the problem for me and some other people

yuxuan-wang-tj, 2021-11-25:
@niemiaszek I have replaced darnet.py in libdarnet with the latest version, and I have changed the class DETECTION in darnet_zed.py to new format. However, The problem still exists .

niemiaszek, 2021-11-25:
Do you use [Alexey darknet repo](https://github.com/AlexeyAB/darknet) for build, libs etc. or do you use the one provided in this repo? Darknet in zed-yolo repo has very old code in it so it might not work with the structure from my PR, didn't test it. I built darknet with instructions from Alexey repo and then changed paths in this zed-yolo repo to match Darknet. I have done this installation month ago and it was working with no problem for me

yuxuan-wang-tj, 2021-11-25:
@niemiaszek Thank you very much. I have solved the problem after I followed the method you said. I truly appreciate your timely help.

niemiaszek, 2021-11-25:
@yuxuan-wang-tj You are welcome 😄