stereolabs/zed-yolo #50: Tracking objects detected with YOLO

Jordi-Gene-Mola, 2021-05-18:
Is it possible to use the object tracking function (integrated in the ZED SDK) to track the objects detected with YOLO?

Sujay-Mhaske, 2021-10-30:
Okay , Thank you for detailed information.
I guess i found to get bounding box on custom data here https://github.com/niemiaszek/zed-yolo/blob/master/zed_python_sample/darknet_zed.py  , it is in def get_object_depth(depth, bounds): the thing is i dont know how to compile darknet can you help me do it via anydesk it i am getting dll errors.

niemiaszek, 2021-10-31:
I am currently away from home - only have my laptop with me. I have not built darknet with windows for a very long time - last time I did it was 2 years ago. For now I would suggest you to use [this installation method](https://github.com/AlexeyAB/darknet#how-to-compile-on-windows-using-cmake) from alexeyab darknet repo. It requires you to get some build tools like visual studio compiler (for compiling on windows) and cmake (to set up build to your configuration). Also remember to download opencv binaries, just as specified [here](https://github.com/AlexeyAB/darknet#requirements-for-windows-linux-and-macos). If you get some problems I can try to help you.

Sujay-Mhaske, 2021-10-31:
Thank you very much , I will try

Sujay-Mhaske, 2021-11-22:
Why are there two rectangles here ?

cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
                              (x_coord + x_extent + thickness, y_coord + (18 + thickness*4)),
                              color_array[detection[3]], -1)
 cv2.putText(image, label + " " +  (str(distance) + " m"),
                            (x_coord + (thickness * 4), y_coord + (10 + thickness * 4)),
                            cv2.FONT_HERSHEY_SIMPLEX, 0.5, (255, 255, 255), 2)
 cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
                              (x_coord + x_extent + thickness, y_coord + y_extent + thickness),
                              color_array[detection[3]], int(thickness*2))

AND in the code it is using .weights file for detection I wanted to use .pt file how should i do that ?

Sujay-Mhaske, 2021-11-22:
Why are there two rectangles here ?

cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
(x_coord + x_extent + thickness, y_coord + (18 + thickness4)),
color_array[detection[3]], -1)
cv2.putText(image, label + " " + (str(distance) + " m"),
(x_coord + (thickness * 4), y_coord + (10 + thickness * 4)),
cv2.FONT_HERSHEY_SIMPLEX, 0.5, (255, 255, 255), 2)
cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
(x_coord + x_extent + thickness, y_coord + y_extent + thickness),
color_array[detection[3]], int(thickness2))

AND in the code it is using .weights file for detection I wanted to use .pt file how should i do that ?

closed 2022-04-26 by github-actions[bot]