stereolabs/zed-yolo #52: ?

Sujay-Mhaske, 2021-11-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Why are there two rectangles in darknet_zed.py?

cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
(x_coord + x_extent + thickness, y_coord + (18 + thickness4)),
color_array[detection[3]], -1)
cv2.putText(image, label + " " + (str(distance) + " m"),
(x_coord + (thickness * 4), y_coord + (10 + thickness * 4)),
cv2.FONT_HERSHEY_SIMPLEX, 0.5, (255, 255, 255), 2)
cv2.rectangle(image, (x_coord - thickness, y_coord - thickness),
(x_coord + x_extent + thickness, y_coord + y_extent + thickness),
color_array[detection[3]], int(thickness2))

AND in the code it is using .weights file for detection I wanted to use .pt file how should i do that ?

### Steps to Reproduce

NA

### Expected Result

NA

### Actual Result

NA

### ZED Camera model

ZED2i

### Environment

```shell
Windows 10
```

niemiaszek, 2021-11-23:
Hm, those rectangles seems to be just detection bounding boxes with description. This code is using darknet framework to run yolo - it's native framework for yolo models. With .pt I guess you want to use pytorch model - this code is not intended to work with pytorch. There is a [pytorch example](https://www.stereolabs.com/docs/pytorch/) for Zed, but if you want to use YOLO just use .weights file.

Sujay-Mhaske, 2021-11-23:
Okay , thanks

closed 2021-11-23 by Sujay-Mhaske

Sujay-Mhaske, 2021-11-26:
> Hm, those rectangles seems to be just detection bounding boxes with description. This code is using darknet framework to run yolo - it's native framework for yolo models. With .pt I guess you want [...]

Okay , thanks . But i wanted to use custom detection by training yolov3 or yolov5 , .pt files how should i do that ? I want to get custom object height and width , which seems not to be in pytorch program.

Can this program https://github.com/stereolabs/zed-examples/blob/b0ad7a91de4b64dbe3296a5bb5ae802ec1913a20/object%20detection/custom%20detector/cpp/tensorrt_yolov5/gen_wts.py create .weights file . I will just change the extension from .wts to .weights as both are weight files and then use in zes-yolo program ?

niemiaszek, 2021-11-28:
So what is the base model that you want to use? I believe that ported yolov4 has tools to convert weights from and to darknet format. I think, that changing the extension might not be enough, because these two frameworks might have completely different format of weights. For yolov4 (trained in darknet framework, where you get .weights) you might go with [this sample](https://github.com/stereolabs/zed-examples/tree/master/object%20detection/custom%20detector/cpp/opencv_dnn_yolov4), where inference is done in opencv dnn, and with yolov5 from ultralitycs (in pytorch, so model is saved as .pt) you can go with [this sample](https://github.com/stereolabs/zed-examples/tree/master/object%20detection/custom%20detector/cpp/tensorrt_yolov5), where you convert pytorch model to tensorRT engine. As I look now these 2 samples are well prepared and more recent than this one, zed-yolo, so you should base on them. Then training your custom detector should be done directly in [yolov4](https://github.com/AlexeyAB/darknet) or [yolov5](https://github.com/ultralytics/yolov5). These examples give you 3D bboxes, so it shouldn't be that hard to approximate height or width

Sujay-Mhaske, 2021-11-29:
okay , thank you very much.